CasualCodersProjects/DS4-Type-C #1: JDS-011 Board pictures

IcicaSora, 2022-03-02:
https://drive.google.com/drive/folders/1vvhLrmgItue2Wch3YJ4nsu_PF8t5DVdx?usp=sharing

While I don't plan to mod my controller, good luck with your project :)

12ian3, 2022-03-02:
Those photos are fantastically useful, and give me the board outline. If you have access to a multimeter, would you be able to get me a pinout on the FPC Connector, LED, and USB? I don't feel like trying to guess and figure out traces from photos.
Biggest things would be 5v, GND, Data +, Data -. That would net a useful board, just without the LED. If you can also trace back the LED pins, that'd be everything I need to spin up a fully functional JDS-011 board.

Here's a photo with MicroUSB's pinout to help with that.
[image]

Thanks!

chand1012, 2022-03-03:
@IcicaSora thank you for the images! Closing issue.

IcicaSora, 2022-03-04:
Apparently [this](https://github.com/HDR/DualShock4-USB-C) already did the handy work (no schematics tho, only gerber, might still be useful).

IcicaSora, 2022-03-04:
there're 2 pins that are connected to each other but are not connected to gnd or vcc, not sure what they're for

[image]

12ian3, 2022-03-04:
Yeah. I based 001 off that, however it's not correct for 011. Note that 011 only has 12 pins on the fpc connector. It may have the same pinout, just two pins shorter, but I can't say for sure without measuring anything.

12ian3, 2022-03-04:
> there're 2 pins that are connected to each other but are not connected to gnd or vcc, not sure what they're for
> 
> [image]

Thank you for your work! I've modified the 001 board to include pads for the 011 board. Hopefully that works. I really like the idea of dual boards, as it reduces overhead and makes it more versatile. I'll get that pushed to the repo soon™.

closed 2022-03-04 by 12ian3

IcicaSora, 2022-03-05:
Have another question: what are the use of the 0 ohm resistors on R3 and R4, and could I just bridge them without using any resistors?

12ian3, 2022-03-06:
> 

I'll assume that's for the 001/011 board. Those are impedance matching resistors for those who think that's truly necessary. I don't see it as necessary, so you should use 0Ω resistors for the most professional/cleanest build. They're like 2 cents each. You could bridge them together if you really need to though.