CasualCodersProjects/DS4-Type-C #4: Question about 5.1kOhm pull-down resistors for CC1 and CC2

CSab6482, 2022-03-30:
I was planning on removing the micro-USB port from my DS4 board and soldering in a USB-C connector (similar to [this video](https://youtu.be/fk7J3LPLqRE) and [this one](https://youtu.be/-FndDPk_CGY)), but that would require that I offset my LED, and I wanted to retain its original position and functionality.

When I found this project, I figured it was a much better fit. Unfortunately, it's only cost effective to make multiple boards, and I only have one controller with a JDS-030 board. In which case, I am eagerly awaiting the opening of the tindie shop with pre-built configurations.

However, my question relates to the ability to use this port with USB-C to USB-C cables. My understanding is that in order to do this, pins A5 and B5 need to be connected to ground with a 5.1kOhm pull-down resistor. Do these boards have connection points for these resistors? For example, [this image](https://github.com/12ian3/DS4-Type-C/blob/main/040%20055/040%20055_Front.png) of the 040/055 board has six spots for resistors to be soldered onto, but what is the resistance measurement of these resistors, and do any of them connect CC1 and CC2 to ground? I was not able to find any recommended resistance levels on the [parts section](https://github.com/12ian3/DS4-Type-C#parts).

In any case, I found that RG1608P-512-D-T5 resistors are the correct 0603 size and they are rated for 5.1kOhms. Would it be better to update the parts section to specify a certain resistance, or will different boards require different resistances?

CSab6482, 2022-03-30:
I found [this similar project](https://youtu.be/jyw3BSlLAMU?t=64) for the JDS-055 board (timestamped when the creator mentions the USB-C to USB-C functionality), maybe certain parts of both projects could be merged together? This design only uses one side of the PCB.

12ian3, 2022-03-30:
You'll find these resistors on all 3 of my designs. Two of them are indeed 5.1k pull downs to ground on CC1 and CC2 of the USB C port. The other two are 0 ohm resistors which are unnecessary, but can allow for impedance matching. When I started this project, I mistook the common mode choke for a resistor network which I assumed was for impedance matching, hence the zero ohm resistors. The 040/05X board is the exception here, since it contains 4 zero ohm resistors. Only two are designed to be populated though.

CSab6482, 2022-03-30:
Gotcha, is there any way for me to know which pads should have the zero ohm resistors and which ones should have the 5.1k ones? I'm still a bit newbish to reading schematics and gerber files and all that.

12ian3, 2022-03-30:
You're correct for 040/055. It may be different for the other boards. I can possibly make some modifications to ensure that R1 and R2 are always the 5.1k resistors and that 3-4(6) are 0 ohms. I won't commit to that just yet though.

CSab6482, 2022-03-30:
Thank you so much for that explanation, I'll keep an eye out for the other boards and I completely understand if it is not possible to make all the resistor locations line up.

closed 2022-03-30 by CSab6482