CasualCodersProjects/DS4-Type-C #6: Board 011 does not have the correct pinout

12ian3, 2022-04-07:
As with 030, the pinout is incorrect. 5 and 6 should be VCC, and 8 should be the Annode of the RGB LED

12ian3, 2022-04-18:
Should've been fixed last night with the latest push. Will leave this issue open until confirmed.

JustinWalravens, 2022-08-03:
Hello,
The problem is fix ? 
thanks

JustinWalravens, 2022-08-03:
Besides, I'm having trouble finding the right reference for the 011 LED.
 Do you have references?
 Thanks in advance.

12ian3, 2022-08-03:
> Hello, The problem is fix ? thanks

JDS-011 has been working since commit 680932b3be2552280a675500a78e0a35e2cc8f59.
The LED actively in use can be found at the bottom of the readme page, with a link to LCSC where there are presently 22,000 in stock. Should you wish to go back to the RHOM LEDs, that is under the RHOM-LEDs branch, though I cannot guarntee function, and also has no support.

closed 2022-08-03 by 12ian3

JustinWalravens, 2022-08-03:
OK, thanks 
 What I didn't understand was that the size of the link LEDs, are much smaller than the original LED. 
 But so if I understand correctly the footprints on the new pcb are the size of the new LED.