CasualCodersProjects/DS4-Type-C #8: Board Question

KozmikNano, 2022-08-11:
Sorry if this is a dumb question but, which Board is considered the best for playing with USB and charging?

12ian3, 2022-08-11:
Hi there,

I don't understand your question. Would you be able to rephrase it for me?

Thanks!

KozmikNano, 2022-08-11:
Out of the list of boards that works for this project, which one is the best? Are they all the same?

12ian3, 2022-08-11:
Each board is unique to a specific controller. They are all the same general schematic, however certain shapes and pinouts are different. Otherwise, they all work the same.

You'll have to open your controller and figure out which of the 5 different models of boards your controller uses, then fab the correct replacement.

KozmikNano, 2022-08-11:
Ok, thanks!

closed 2022-08-11 by KozmikNano